Frogging-Family/nvidia-all #243: Compiling open kernel modules fails on linux-zen and other non default kernels

Alex-MacLean, 2024-06-01:
### **The Issue**
Compiling the open kernel modules fails on non default kernels such as linux-zen due to /usr/src/linux not being present, with linux-zen the appropriate symlink is /usr/src/linux-zen.

### **My Workaround**
I was able to workaround this issue by creating a symlink from /usr/src/linux pointing to /usr/src/linux-zen. This allows the script to properly compile the nvidia open kernel modules as they are hardcoded to expect the kernel headers to be at 
/usr/src/linux

### **Possible Solutions**
This issue could be solved by temporarily linking the detected or chosen headers directory to /usr/src/linux inside the shell instance or by using temporary symlinks which would be less ideal, but would still work. If /usr/src/linux is hardcoded in the open kernel modules script then it would be possible to detect the installed headers or ask the user if they are using a different kernel and change the path based on what is detected or what the user inputs.

Tk-Glitch, 2024-06-04:
You only need to have the stock kernel headers *installed* at build time. DKMS will take care of your other kernels just fine when installing the modules. We used to have a check for headers that we had to remove because of custom kernel BS. The sensible solution would be to add the `linux-headers` package to make dependencies and be done with it.

I have abandoned the idea of adapting to whatever custom kernel is installed as it's simply not possible. Between "broken" headers, kernel naming inconsistencies and other fun new breakage every other day it's a PITA.

Alex-MacLean, 2024-06-11:
That makes a lot of sense. Accounting for kernel and bs people add to their sometimes broken custom kernels and headers really doesn't sound worth it when nothing is gained. I am not very experienced in kernel space development so that just kind of flew over my head. I'll close this issue since there is no reason for this issue to exist anymore.

closed 2024-06-11 by Alex-MacLean

Tk-Glitch, 2024-06-11:
I understand the pain as a user too though :D I'll think about the best way to handle this.

Alex-MacLean, 2024-06-11:
I'll just keep using my symlink workaround since I don't feel like having another headers package installed and the linux-zen headers tend to work properly as linux-zen is an "official" arch kernel. Maybe customization.cfg would be a good place for options for custom headers since it's out of the way for most users and is there for people who are looking for it.

Tk-Glitch, 2024-07-08:
Possibly in an acceptable fashion after https://github.com/Frogging-Family/nvidia-all/commit/59d6bef220c0d17adef557a39b027822a07dd5c0