Frogging-Family/nvidia-all #246: unresolvable dependencies

Dunravin, 2024-06-07:
Hello, I had this working previously but totally borked my setup and ended up reinstalling arch but I can't for the life of me figure it out this time.

```
loading packages...
resolving dependencies...
warning: cannot resolve "lib32-zlib", a dependency of "lib32-nvidia-utils-tkg"
warning: cannot resolve "lib32-libglvnd", a dependency of "lib32-nvidia-utils-tkg"
warning: cannot resolve "lib32-mesa", a dependency of "lib32-nvidia-utils-tkg"
warning: cannot resolve "lib32-vulkan-icd-loader", a dependency of "lib32-nvidia-utils-tkg"
warning: cannot resolve "lib32-zlib", a dependency of "lib32-opencl-nvidia-tkg"
:: The following packages cannot be upgraded due to unresolvable dependencies:
lib32-nvidia-utils-tkg  lib32-opencl-nvidia-tkg

:: Do you want to skip the above packages for this upgrade? [y/N]
```

Thanks for any help.

ptr1337, 2024-06-07:
enable multilib

Dunravin, 2024-06-07:
Thank you, I just figured it out.

for anyone else un-comment the multilib section in /etc/pacman.conf

closed 2024-07-02 by Tk-Glitch