Frogging-Family/nvidia-all #251: 555.58.02 Released

1nikolas, 2024-07-01:
Driver 555.58.02 (bugfix) has been released

https://www.nvidia.com/Download/driverResults.aspx/228410/en-us/

Thanks

pipes80-collab, 2024-07-01:
Good, but this driver solve the dims problem with 6.10-rc?

ptr1337, 2024-07-01:
> Good, but this driver solve the dims problem with 6.10-rc?

There is no problem with 6.10 since a while. Latest 555.58 works fine and also latest 550.90.

pipes80-collab, 2024-07-01:
Not for me, I can't load dims modules with 6.10-rc2

ptr1337, 2024-07-04:
> Not for me, I can't load dkms modules with 6.10-rc2

That appears to be an issue in your configuration or equal. We are building the NVIDIA Module since the first release of 555 and also several other users are using it without any problems.

Tk-Glitch, 2024-07-08:
Closed with https://github.com/Frogging-Family/nvidia-all/commit/4d2be0fa0937daa69d6bb7af1e322bba0fb358fe

closed 2024-07-08 by Tk-Glitch

pipes80-collab, 2024-07-09:
Maybe, because I have no problem with Linux kernel 6.9.8 from arch repository