Frogging-Family/nvidia-all #262: Open Source Kernel Module 560.35.03 not functional on Kernel 6.10.11

Alex-MacLean, 2024-09-16:
I just updated my system today and after rebooting my desktop environment would not initialize. The display manager worked fine, but the desktop would not initialize. Downgrading back to 6.10.10 resolved the issue. The kernel module compiled successfully, DKMS installed it correctly, and mkinitcpio created the initramfs successfully, but my desktop environment and/or Wayland failed to initialize correctly and left me at a blank screen. Switching to TTY3 worked allowed me to downgrade my kernel.

Running "sudo dmesg | grep nvidia" showed nothing out of the ordinary and lsmod showed all nvidia modules as present so likely some update in the kernel changed something Nvidia's kernel modules uses when initializing the display server.

This may be more of a feature request, but I would appreciate a patch for the new kernel if at all possible as Nvidia tends to take their time with fixing their drivers. 

For now I'll stay on 6.10.10, not a huge deal, but I like keeping my system as up to date as possible. 

I'm unsure if this is the right place for a kernel module compatibility issue, but it is potentially possible a patch can be made if the root cause and a viable solution are found.

solarisfire, 2024-09-21:
Are you running an Intel CPU with an iGPU on it?

Alex-MacLean, 2024-09-25:
> Are you running an Intel CPU with an iGPU on it?

No, I have an AMD Ryzen 7 3700x

solarisfire, 2024-10-02:
Looks like this is also broken with Kernel 6.11.1

Apparantly Nvidia has a patch... Can nvidia-all include this?

https://gitlab.archlinux.org/archlinux/packaging/packages/linux/-/issues/80

Alex-MacLean, 2024-10-05:
The driver works properly when enabling nvidia-drm modeset and fbdev by setting the nvidia_drm.modeset=1 and nvidia_drm.fbdev=1 kernel parameters and in /etc/modprobe.d/nvidia.conf adding "option nvidia-drm modeset=1 fbdev=1". From the issue mentioned above it looks like a simpledrm issue and these options essentially disable simpledrm and use nvidia-drm exclusively

solarisfire, 2024-10-05:
I have the kernel parameters set as nvidia_drm.modeset=1 nvidia_drm.fbdev=1

I also have tried setting it in /etc/modprobe.d/nvidia.conf as:

<pre>
options nvidia_drm modeset=1
options nvidia_drm fbdev=1
</pre>

Installing kernel 6.11.1 with driver 560.35.03, and I just get a black screen when logging in as wayland starts and kde plasma loads. SDDM works fine however.

Tk-Glitch, 2024-12-01:
We're doing this on both full prop and kernel-open packages now, so I guess we can close this.

closed 2024-12-01 by Tk-Glitch