Frogging-Family/nvidia-all #264: Error: failed to commit transaction (conflicting files)

mysteryx93, 2024-09-22:
Trying to revert to 555 drivers as I'm having a few issues in 560 recently.  The install script isn't working.
```
makepkg -si

Package (14)             Old Version    New Version    Net Change

egl-wayland              4:1.1.16-1.1                    -0.10 MiB
lib32-nvidia-utils       560.35.03-1.1                 -205.01 MiB
lib32-opencl-nvidia      560.35.03-1.1                  -19.04 MiB
nvidia-dkms              560.35.03-3.1                  -80.06 MiB
nvidia-settings          560.35.03-1.1                   -1.54 MiB
nvidia-utils             560.35.03-3.1                 -652.24 MiB
opencl-nvidia            560.35.03-3.1                  -28.10 MiB
lib32-nvidia-utils-tkg                  555.58.02-257   195.65 MiB
lib32-opencl-nvidia-tkg                 555.58.02-257    18.73 MiB
nvidia-dkms-tkg                         555.58.02-257    68.91 MiB
nvidia-egl-wayland-tkg                  555.58.02-257     0.08 MiB
nvidia-settings-tkg                     555.58.02-257     1.67 MiB
nvidia-utils-tkg                        555.58.02-257   616.46 MiB
opencl-nvidia-tkg                       555.58.02-257    21.92 MiB

Total Installed Size:  923.44 MiB
Net Upgrade Size:      -62.66 MiB

:: Proceed with installation? [Y/n] y
(7/7) checking keys in keyring                                    [------------------------------------] 100%
(7/7) checking package integrity                                  [------------------------------------] 100%
(7/7) loading package files                                       [------------------------------------] 100%
(7/7) checking for file conflicts                                 [------------------------------------] 100%
error: failed to commit transaction (conflicting files)
nvidia-utils-tkg: /usr/lib/libnvidia-egl-gbm.so exists in filesystem (owned by egl-gbm)
nvidia-utils-tkg: /usr/lib/libnvidia-egl-gbm.so.1 exists in filesystem (owned by egl-gbm)
nvidia-utils-tkg: /usr/share/egl/egl_external_platform.d/15_nvidia_gbm.json exists in filesystem (owned by egl-gbm)
Errors occurred, no packages were upgraded.
==> WARNING: Failed to install built package(s).
-> exit cleanup done
```

Operachi061, 2024-09-22:
try to remove egl-gbm package

ptr1337, 2024-09-22:
egl-gbm is now dependecy of arch's nvidia-utils package. Should be also adjusted here.

Operachi061, 2024-09-22:
idk he just can make ``pacman -Rcns nvidia-utils`` cuz this can remove egl-gbm too and then install this nvidia tkg drivers

closed 2024-09-22 by Tk-Glitch