Frogging-Family/nvidia-all #267: Error:

vitox420, 2024-10-04:
I get the Error message: ==> ERROR: A failure occurred in build(). Aborting..

- 560 series
- dmks
- Manjaro 6.10 kernel

This logs might or might not help:
- date: version.mk: No such file or directory
- make[1]: *** [Makefile:89: modules] Error 2
- make: *** [Makefile:59: modules] Error 2

maybe kernel 6.10 aint supported yet?

Thank you for your work. I just discovered this git and had enough problems with nvidia driverz. Looking forward.

werelell, 2024-10-20:
im having the same issue with the nvidia 470 driver

pcboomer33, 2024-10-21:
I'm having the same issue. It doesn't just occur with the 560 series.

damachine, 2026-03-01:
Please reopen if this is still relevant.

closed 2026-03-01 by damachine