Frogging-Family/nvidia-all #275: Potential issue with kernel 6.12

GenocideStomper, 2024-11-30:
When updating to kernel 6.12 (either through arch repos or compiling linux-tkg), KDE Plasma doesn't start and booting is stuck. There were no issues with dkms during installation.

Uninstalling Frogging's nvidia-all and reverting to arch repos for nvidia solves the issue, even though they're the same driver version.

Only discussion I found was here: https://old.reddit.com/r/archlinux/comments/1gykccg/psa_nvidia_users_612_causes_breakage_on_565/lz9htkp/

My system:
endeavourOS
KDE Plasma X11
GTX 1080
Proprietary driver 565.57.01

jnines, 2024-12-01:
Maybe it's a bit of a circular issue?  If you don't already have a 6.12.x kernel installed, the PKGBUILD won't install/patch the driver for 6.12.  If you then install and run 6.12.x it'll tank Plasma because the driver isn't patched.  

The only thing you could do at that point is, with a 6.12.x kernel installed, either jump to a tty and run the PKGBUILD from there, or load into another kernel and run the PKGBUILD.  Before the commit here I just checked if the driver was 565* and patched it.  It didn't cause any issues with 6.11 but I can't speak for kernels previous to that.

GenocideStomper, 2024-12-01:
Oh that is probably the case. I'll close this for now

itouchdennis, 2024-12-03:
I think I have the same issue, 6.12 with 565 and open drivers brakes the system for me with the froggi install.

I tried the workaround mentioned, installing 6.12 reboot, tty and rebuild the packages (I pulled the latest changes). 

Reboot and hanging again after the mount points. 
Do I need to remove all nvidia packages before?

Tk-Glitch, 2024-12-03:
I was hopeful https://github.com/Frogging-Family/nvidia-all/commit/3cd89c766ab34addb355f2d9fecdadf2d9a4456f would fix it. That's annoying, I don't have access to a GPU supported by the open-kernel drivers.

itouchdennis, 2024-12-03:
Any chances I can test it for you? I know not the best solution..

itouchdennis, 2024-12-03:
I was also searching on the nvidia 565 forum about any hints, it was mentioned this commit:

https://github.com/NVIDIA/open-gpu-kernel-modules/issues/712
here
https://forums.developer.nvidia.com/t/565-release-feedback-discussion/310777/206

Is this by any chance helpful? Sry, I am not that deep into.

jnines, 2024-12-03:
> I was hopeful [3cd89c7](https://github.com/Frogging-Family/nvidia-all/commit/3cd89c766ab34addb355f2d9fecdadf2d9a4456f) would fix it. That's annoying, I don't have access to a GPU supported by the open-kernel drivers.

In the case of the 6.12 patch (can't speak to others), `kernel-dkms` has to be `kernel-open` for the open module.

Fxzzi, 2024-12-05:
you should check out what the official arch nvidia package has done, seems to work fine for me on 6.12. also new update btw: https://www.nvidia.com/en-us/drivers/details/237598/

padoruuuu, 2024-12-06:
as of now things seem to be working on 6.12.

itouchdennis, 2024-12-07:
Works like a charm now!

closed 2026-03-01 by damachine