Frogging-Family/nvidia-all #287: GCC 15

damachine, 2025-05-02:
hi , we need a update to compile with gcc15. [https://github.com/CachyOS/CachyOS-PKGBUILDS/commit/60b77fa3e5e6bf5446bbeec0a10ab263fd9dc9da](https://github.com/CachyOS/CachyOS-PKGBUILDS/commit/60b77fa3e5e6bf5446bbeec0a10ab263fd9dc9da)](url)
thx

lwierez, 2025-05-06:
Make has a way for you to choose what version of GCC you want to use.

First, install the version you need with
`pacman -S gccXX`
XX being the major, so here you want the package gcc14

Then, you can use the env var CC to choose the compiler you just installed
`CC=gcc-14 makepkg -si`

Glog78, 2025-05-07:
> Make has a way for you to choose what version of GCC you want to use.
> 
> First, install the version you need with `pacman -S gccXX` XX being the major, so here you want the package gcc14
> 
> Then, you can use the env var CC to choose the compiler you just installed `CC=gcc-14 makepkg -si`
> 
> We might indeed need a fix in the repo, but this should be a working fix for now 🙂

I am not sure if this is really a solution ... is CC saved and stored so it gets used in the DKMS builds after too ? I am asking out of curiosity . When i downgraded gcc all was fine ... when i used CC=gcc-14 only for one kernel dkms could build a kernel module ...

damachine, 2025-05-07:
hi. arch ist update gcc to v15xxx. i think need to PATCH CFLAGS: Set -std=gnu17 for all compilation flags
 
 GCC15 uses C23 by default, which breaks a lot of compatibility with the
 kernel and this module. downgrade to gcc 14 works while kernel also n gcc 14.

ltrii, 2025-05-07:
I can confirm this works for anyone getting headaches. Not sure if it's relevant but I use ccache so I cleared that out (`ccache -C`) before building cause I had not cleared it in a bit. Using the latest kernel (6.14.5) building the DKMS open kernel modules with ` CC=gcc-14 makepkg -si` was a working workaround for me.

damachine, 2025-05-07:
think kernel must also compile with this trick before. than it should work. but most have gcc15 now since last update.

jamesb6626, 2025-05-08:
Not tried it, but searching the internet suggests the concerns of @Glog78 are correct, in that it looks like `CC=gcc-14` maybe won't be saved by dkms. I had to patch `CC=gcc-14` into the `MAKE[0]` line of `dkms.conf`. Otherwise, I suspect kernel updates will fail to compile this. This will also make `gcc14` a dependency rather than a build-time dependency for the package, until version 15 is compatible.

FuzzyQuils, 2025-05-10:
> Not tried it, but searching the internet suggests the concerns of [@Glog78](https://github.com/Glog78) are correct, in that it looks like `CC=gcc-14` maybe won't be saved by dkms. I had to patch `CC=gcc-14` into the `MAKE[0]` line of `dkms.conf`, so probably this can just be patched in this repo right? Otherwise, I suspect kernel updates will fail to compile this. This will also make `gcc14` a dependency rather than a build-time dependency for the package, until version 15 is compatible.
> 
> (for other stuck, until this gets patched, I edited `/usr/src/nvidia-*/dkms.conf` followed by `dkms autoinstall`)

Another option that works for `nvidia-open` is to add this line underneath line 89 in `/usr/src/nvidia-*/kernel-open/Kbuild`:
`EXTRA_CFLAGS += -std=gnu17`

Works for conftest as well.

What I did: 
- install `nvidia-all` with `CC=gcc-14`. conftest throws a warning but passes anyway.
- after initially installing the packages, navigate to that path and edit Kbuild.
- run `sudo dkms autoinstall` (or the equivalent for ALL installed kernels, forgot how)
- reboot.

jamesb6626, 2025-05-10:
> Another option that works for nvidia-open is to add this line underneath line 89 in /usr/src/nvidia-*/kernel-open/Kbuild:
EXTRA_CFLAGS += -std=gnu17

Interesting. If dkms is installing that correctly with only that flag, then there isn't even a need for a `gcc14` dependency. Looks like that's exactly what the CachyOS patch is doing too, as a workaround for `gcc15` compatibility. There should be a note to change it when nvidia is compatible with a higher std, but looks like `-std=gnu17` is all that's needed. Just patch that in then the initial build and dkms should work fine with no extra work hopefully.

Tk-Glitch, 2025-05-13:
Fixed with https://github.com/Frogging-Family/nvidia-all/commit/99751f574bce22a229a2f338f6444b1866498724

closed 2025-05-13 by Tk-Glitch

damachine, 2025-05-13:
sorry but for me compiling fails.
```make[5]: *** [/usr/lib/modules/6.14.6-273-linux-tkg/build/scripts/Makefile.build:207: nvidia/nv-pat.o] Fehler 1
make[5]: *** [/usr/lib/modules/6.14.6-273-linux-tkg/build/scripts/Makefile.build:207: nvidia/nv-mmap.o] Fehler 1
make[5]: *** [/usr/lib/modules/6.14.6-273-linux-tkg/build/scripts/Makefile.build:207: nvidia/nv.o] Fehler 1
nvidia/nv-backlight.c: In Funktion »nv_get_tegra_brightness_level«:
nvidia/nv-backlight.c:41:10: Fehler: Implizite Deklaration der Funktion »get_backlight_device_by_name«; meinten Sie »backlight_device_get_by_name«? [-Wimplicit-function-declaration]
   41 |     bd = get_backlight_device_by_name(nvl->backlight.device_name);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |          backlight_device_get_by_name
nvidia/nv-backlight.c:41:8: Fehler: Zuweisung an »struct backlight_device *« von »int« wandelt eine Zahl in einen Zeiger um, ohne explizite Typkonvertierung [-Wint-conversion]
   41 |     bd = get_backlight_device_by_name(nvl->backlight.device_name);
      |        ^
nvidia/nv-backlight.c: Auf höchster Ebene:
nvidia/nv-backlight.c:56:23: Warnung: kein vorheriger Prototyp für »nv_set_tegra_brightness_level« [-Wmissing-prototypes]
   56 | NV_STATUS NV_API_CALL nv_set_tegra_brightness_level
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-backlight.c: In Funktion »nv_set_tegra_brightness_level«:
nvidia/nv-backlight.c:66:8: Fehler: Zuweisung an »struct backlight_device *« von »int« wandelt eine Zahl in einen Zeiger um, ohne explizite Typkonvertierung [-Wint-conversion]
   66 |     bd = get_backlight_device_by_name(nvl->backlight.device_name);
      |        ^
make[5]: *** [/usr/lib/modules/6.14.6-273-linux-tkg/build/scripts/Makefile.build:207: nvidia/nv-backlight.o] Fehler 1
make[4]: *** [/usr/lib/modules/6.14.6-273-linux-tkg/build/Makefile:2006: .] Fehler 2
make[3]: *** [/usr/lib/modules/6.14.6-273-linux-tkg/build/Makefile:251: __sub-make] Fehler 2
make[3]: Verzeichnis „/tmp/makepkg/nvidia-open-dkms-tkg/src/NVIDIA-kernel-module-source-575.51.02/kernel-open“ wird verlassen
make[2]: *** [Makefile:251: __sub-make] Fehler 2
make[2]: Verzeichnis „/usr/lib/modules/6.14.6-273-linux-tkg/build“ wird verlassen
make[1]: *** [Makefile:140: modules] Fehler 2
make[1]: Verzeichnis „/tmp/makepkg/nvidia-open-dkms-tkg/src/NVIDIA-kernel-module-source-575.51.02/kernel-open“ wird verlassen
make: *** [Makefile:59: modules] Fehler 2
==> FEHLER: Ein Fehler geschah in build().
    Breche ab...
  -> exit cleanup done

ltrii, 2025-05-13:
To follow up on what @damachine got I pulled in the changes and new patch, made sure I was up to date across the board on my machine, and ran `makepkg -si` against a clean pull. Got the same exact error. Adding `CC=gcc-14` before the command worked once more (again I have those binaries explicitly installed). The build does warn about the compiler version mismatch between the kernel and build during the process, but everything works fine once the package is built.

jamesb6626, 2025-05-14:
Yep, still broken, gotta reopen. Is there a good way to pass CFLAGS through in the `dkms.conf`? Would `EXTRA_CFLAGS` or something there get passed through? I wanna do a hard test, to see if it's just not getting applied in the right place, or if `-std=gnu17` is not solving the problem. Maybe `gcc-14` is required after all, but let's see.

jamesb6626, 2025-05-14:
It is very strange that `-std=gnu17` doesn't work, it seems it works for someone in the nvidia repo:
https://github.com/NVIDIA/open-gpu-kernel-modules/issues/814
https://github.com/NVIDIA/open-gpu-kernel-modules/pull/841

Maybe a `gcc14` dependency is the way to go for now, until upstream fixes this properly.

jamesb6626, 2025-05-14:
Okay, seems that `EXTRA_CFLAGS += -std=gnu17`, as per @FuzzyQuils suggestion, works for me, which makes it even stranger that the patch doesn't work. So yeah, `gcc-14` not required, but no idea why the patch doesn't work, despite getting seemingly getting applied.

Glog78, 2025-05-16:
Thanks alot for fixing it

lwierez, 2025-05-23:
> > Make has a way for you to choose what version of GCC you want to use.
> > First, install the version you need with `pacman -S gccXX` XX being the major, so here you want the package gcc14
> > Then, you can use the env var CC to choose the compiler you just installed `CC=gcc-14 makepkg -si`
> > We might indeed need a fix in the repo, but this should be a working fix for now 🙂
> 
> I am not sure if this is really a solution ... is CC saved and stored so it gets used in the DKMS builds after too ? I am asking out of curiosity . When i downgraded gcc all was fine ... when i used CC=gcc-14 only for one kernel dkms could build a ( !! one my current ) kernel module ...

Sorry, it's been a while and the issue is closed, but this is an interesting thing about Linux:
- If you use `FLAG=value command`, FLAG will only be set at VALUE for the command, outside or after the execution of this command, this flag change basically does not exists
- If you use `export FLAG=VALUE`, FLAG will be set at value for your session and user
- If you want to set a flag system wide, or set a flag outside of your session (so your system boots with this flag), there are different ways but it's more complex

Source: https://wiki.archlinux.org/title/Environment_variables