Frogging-Family/nvidia-all #288: Fails to install on Kernel 6.14.4

Martmists-GH, 2025-05-06:
Install log:

[/var/lib/dkms/nvidia/575.51.02/build/make.log](https://github.com/user-attachments/files/20062483/log.txt)

Martmists-GH, 2025-05-06:
Build log (open, truncated because it was in terminal):

```
nvidia/nv-procfs.c:1359:13: note: in expansion of macro ‘NV_CREATE_PROC_FILE’
 1359 |     entry = NV_CREATE_PROC_FILE("version", proc_nvidia, version, NULL);
      |             ^~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.14.4-zen1-2-zen/build/include/linux/proc_fs.h:111:48: note: expected ‘const struct proc_ops *’ but argument is of type ‘const nv_proc_ops_t *’ {aka ‘const struct file_operations *’}
  111 |                                                const struct proc_ops *,
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-procfs.c: In function ‘nv_procfs_add_gpu’:
././common/inc/nv-procfs-utils.h:73:60: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Wincompatible-pointer-types]
   73 |         __entry = proc_create_data(filename, mode, parent, fops, __data);\
      |                                                            ^~~~
      |                                                            |
      |                                                            const nv_proc_ops_t * {aka const struct file_operations *}
nvidia/nv-procfs.c:1401:13: note: in expansion of macro ‘NV_CREATE_PROC_FILE’
 1401 |     entry = NV_CREATE_PROC_FILE("information", proc_nvidia_gpu, gpu_info,
      |             ^~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.14.4-zen1-2-zen/build/include/linux/proc_fs.h:111:48: note: expected ‘const struct proc_ops *’ but argument is of type ‘const nv_proc_ops_t *’ {aka ‘const struct file_operations *’}
  111 |                                                const struct proc_ops *,
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~
././common/inc/nv-procfs-utils.h:73:60: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Wincompatible-pointer-types]
   73 |         __entry = proc_create_data(filename, mode, parent, fops, __data);\
      |                                                            ^~~~
      |                                                            |
      |                                                            const nv_proc_ops_t * {aka const struct file_operations *}
nvidia/nv-procfs.c:1406:13: note: in expansion of macro ‘NV_CREATE_PROC_FILE’
 1406 |     entry = NV_CREATE_PROC_FILE("registry", proc_nvidia_gpu, registry, nv);
      |             ^~~~~~~~~~~~~~~~~~~
make[5]: *** [/usr/lib/modules/6.14.4-zen1-2-zen/build/scripts/Makefile.build:207: nvidia/nv-dsi-parse-panel-props.o] Error 1
/usr/lib/modules/6.14.4-zen1-2-zen/build/include/linux/proc_fs.h:111:48: note: expected ‘const struct proc_ops *’ but argument is of type ‘const nv_proc_ops_t *’ {aka ‘const struct file_operations *’}
  111 |                                                const struct proc_ops *,
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~
make[5]: *** [/usr/lib/modules/6.14.4-zen1-2-zen/build/scripts/Makefile.build:207: nvidia/nv-cray.o] Error 1
././common/inc/nv-procfs-utils.h:73:60: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Wincompatible-pointer-types]
   73 |         __entry = proc_create_data(filename, mode, parent, fops, __data);\
      |                                                            ^~~~
      |                                                            |
      |                                                            const nv_proc_ops_t * {aka const struct file_operations *}
nvidia/nv-procfs.c:1410:13: note: in expansion of macro ‘NV_CREATE_PROC_FILE’
 1410 |     entry = NV_CREATE_PROC_FILE("power", proc_nvidia_gpu, power, nv);
      |             ^~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.14.4-zen1-2-zen/build/include/linux/proc_fs.h:111:48: note: expected ‘const struct proc_ops *’ but argument is of type ‘const nv_proc_ops_t *’ {aka ‘const struct file_operations *’}
  111 |                                                const struct proc_ops *,
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~
././common/inc/nv-procfs-utils.h:73:60: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Wincompatible-pointer-types]
   73 |         __entry = proc_create_data(filename, mode, parent, fops, __data);\
      |                                                            ^~~~
      |                                                            |
      |                                                            const nv_proc_ops_t * {aka const struct file_operations *}
nvidia/nv-procfs.c:1416:17: note: in expansion of macro ‘NV_CREATE_PROC_FILE’
 1416 |         entry = NV_CREATE_PROC_FILE("exercise_error_forwarding", proc_nvidia_gpu,
      |                 ^~~~~~~~~~~~~~~~~~~
/usr/lib/modules/6.14.4-zen1-2-zen/build/include/linux/proc_fs.h:111:48: note: expected ‘const struct proc_ops *’ but argument is of type ‘const nv_proc_ops_t *’ {aka ‘const struct file_operations *’}
  111 |                                                const struct proc_ops *,
      |                                                ^~~~~~~~~~~~~~~~~~~~~~~
././common/inc/nv-procfs-utils.h:73:60: error: passing argument 4 of ‘proc_create_data’ from incompatible pointer type [-Wincompatible-pointer-types]
   73 |         __entry = proc_create_data(filename, mode, parent, fops, __data);\
      |                                                            ^~~~
[... cut ...]
make[3]: *** [/usr/lib/modules/6.14.4-zen1-2-zen/build/Makefile:251: __sub-make] Error 2
make[3]: Leaving directory '/tmp/tmp.GzAkFDiwup/src/NVIDIA-kernel-module-source-575.51.02/kernel-open'
make[2]: *** [Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.14.4-zen1-2-zen/build'
make[1]: *** [Makefile:140: modules] Error 2
make[1]: Leaving directory '/tmp/tmp.GzAkFDiwup/src/NVIDIA-kernel-module-source-575.51.02/kernel-open'
make: *** [Makefile:59: modules] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
  -> exit cleanup done
```

jamesb6626, 2025-05-06:
I started having a very similar problem, with very similar log output (I _believe_ these very same errors were present, though not in this order - will need to compare my logs, and not sure if the order of errors is deterministic). Might be unrelated though.

For me the problem first occurred after a system upgrade, specifically one which brought me from a linux kernel based on `6.14.3` to `6.14.4`, during an automatic dkms rebuild. Strangely though, the problem seemed to affect my other installed kernel too, a 6.12 version. The dkms install failed for both, as well as on `6.14.5`, as did asking `nvidia-all` to install an older version of the graphics drivers, installing the open/proprietary ones, and installing without dkms. In the end, I had to switch from the nvidia-all packages to `nvidia-dkms` and friends version `570.144-3` through the package manager, which does work.

As a note, my system is an install of garuda, which uses dracut. Again, my problem might not be the same, but if it is, seems it was quite likely introduced on update to kernel `6.14.4`.

Martmists-GH, 2025-05-06:
> still no idea why it affected other kernels though
Yeah, for me it also seemed to affect the LTS kernel.

Ethaks, 2025-05-06:
I'd expect the cause to be #287 — the compile process not working with the driver would make it fail regardless of which kernel version it's built against.

jamesb6626, 2025-05-08:
Can confirm, for me, `gcc` was indeed upgraded to version 15 the very update where dkms first failed. Installed successfully with `gcc14` package, after setting `CC=gcc-14` for dkms (will comment on #287 with what worked), so seems like that's my issue at least.

Tk-Glitch, 2025-05-13:
The GCC issue should be fixed with https://github.com/Frogging-Family/nvidia-all/commit/99751f574bce22a229a2f338f6444b1866498724

Martmists-GH, 2025-05-14:
Still fails: [make.log](https://github.com/user-attachments/files/20197660/make.log)

Martmists-GH, 2025-05-14:
Log from building open driver:

```
 1964 |     num_read = kernel_read(pFile, f_pos, pBuffer, size);
      |                                                   ^~~~
      |                                                   |
      |                                                   NvU64 {aka long long unsigned int}
/usr/lib/modules/6.14.5-zen1-1-zen/build/include/linux/fs.h:3187:59: note: expected ‘loff_t *’ {aka ‘long long int *’} but argument is of type ‘NvU64’ {aka ‘long long unsigned int’}
 3187 | extern ssize_t kernel_read(struct file *, void *, size_t, loff_t *);
      |                                                           ^~~~~~~~
nvidia/os-interface.c: At top level:
nvidia/os-interface.c:2125:23: warning: no previous prototype for ‘os_get_tegra_platform’ [-Wmissing-prototypes]
 2125 | NV_STATUS NV_API_CALL os_get_tegra_platform
      |                       ^~~~~~~~~~~~~~~~~~~~~
nvidia/os-interface.c: In function ‘os_numa_add_gpu_memory’:
nvidia/os-interface.c:2388:15: error: too few arguments to function ‘add_memory_driver_managed’; expected 5, have 4
 2388 |         ret = add_memory_driver_managed(node, segment_base, segment_size, "System RAM (NVIDIA)");
      |               ^~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /usr/lib/modules/6.14.5-zen1-1-zen/build/include/linux/mmzone.h:1475,
                 from /usr/lib/modules/6.14.5-zen1-1-zen/build/include/linux/gfp.h:7,
                 from /usr/lib/modules/6.14.5-zen1-1-zen/build/include/linux/mm.h:7:
/usr/lib/modules/6.14.5-zen1-1-zen/build/include/linux/memory_hotplug.h:312:12: note: declared here
  312 | extern int add_memory_driver_managed(int nid, u64 start, u64 size,
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-vm.c:95:13: error: implicit declaration of function ‘set_memory_array_wb’; did you mean ‘set_pages_array_wb’? [-Wimplicit-function-declaration]
   95 |             set_memory_array_wb(pages, num_pages);
      |             ^~~~~~~~~~~~~~~~~~~
      |             set_pages_array_wb
././common/inc/nv-linux.h: In function ‘nv_is_dma_direct’:
././common/inc/nv-linux.h:1141:9: error: implicit declaration of function ‘dma_is_direct’; did you mean ‘d_is_dir’? [-Wimplicit-function-declaration]
 1141 |     if (dma_is_direct(get_dma_ops(dev)))
      |         ^~~~~~~~~~~~~
      |         d_is_dir
make[5]: *** [/usr/lib/modules/6.14.5-zen1-1-zen/build/scripts/Makefile.build:207: nvidia/nv.o] Error 1
make[5]: *** [/usr/lib/modules/6.14.5-zen1-1-zen/build/scripts/Makefile.build:207: nvidia/nv-vm.o] Error 1
nvidia/nv-dmabuf.c: In function ‘nv_dma_buf_create’:
nvidia/nv-dmabuf.c:1174:6: error: ‘NV_DMA_BUF_EXPORT_ARGUMENT_COUNT’ is not defined, evaluates to ‘0’ [-Werror=undef]
 1174 | #if (NV_DMA_BUF_EXPORT_ARGUMENT_COUNT == 1)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-dmabuf.c:1185:8: error: ‘NV_DMA_BUF_EXPORT_ARGUMENT_COUNT’ is not defined, evaluates to ‘0’ [-Werror=undef]
 1185 | #elif (NV_DMA_BUF_EXPORT_ARGUMENT_COUNT == 4)
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
nvidia/nv-dmabuf.c:1188:8: error: ‘NV_DMA_BUF_EXPORT_ARGUMENT_COUNT’ is not defined, evaluates to ‘0’ [-Werror=undef]
 1188 | #elif (NV_DMA_BUF_EXPORT_ARGUMENT_COUNT == 5)
      |        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[5]: *** [/usr/lib/modules/6.14.5-zen1-1-zen/build/scripts/Makefile.build:207: nvidia/os-interface.o] Error 1
make[5]: *** [/usr/lib/modules/6.14.5-zen1-1-zen/build/scripts/Makefile.build:207: nvidia/nv-pci.o] Error 1
nvidia/nv-backlight.c: In function ‘nv_get_tegra_brightness_level’:
nvidia/nv-backlight.c:41:10: error: implicit declaration of function ‘get_backlight_device_by_name’; did you mean ‘backlight_device_get_by_name’? [-Wimplicit-function-declaration]
   41 |     bd = get_backlight_device_by_name(nvl->backlight.device_name);
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |          backlight_device_get_by_name
nvidia/nv-backlight.c:41:8: error: assignment to ‘struct backlight_device *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
   41 |     bd = get_backlight_device_by_name(nvl->backlight.device_name);
      |        ^
nvidia/nv-backlight.c: At top level:
nvidia/nv-backlight.c:56:23: warning: no previous prototype for ‘nv_set_tegra_brightness_level’ [-Wmissing-prototypes]
   56 | NV_STATUS NV_API_CALL nv_set_tegra_brightness_level
      |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
nvidia/nv-backlight.c: In function ‘nv_set_tegra_brightness_level’:
nvidia/nv-backlight.c:66:8: error: assignment to ‘struct backlight_device *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
   66 |     bd = get_backlight_device_by_name(nvl->backlight.device_name);
      |        ^
make[5]: *** [/usr/lib/modules/6.14.5-zen1-1-zen/build/scripts/Makefile.build:207: nvidia/nv-dmabuf.o] Error 1
make[5]: *** [/usr/lib/modules/6.14.5-zen1-1-zen/build/scripts/Makefile.build:207: nvidia/nv-backlight.o] Error 1
make[4]: *** [/usr/lib/modules/6.14.5-zen1-1-zen/build/Makefile:1995: .] Error 2
make[3]: *** [/usr/lib/modules/6.14.5-zen1-1-zen/build/Makefile:251: __sub-make] Error 2
make[3]: Leaving directory '/tmp/tmp.2h3tBxgCGs/src/NVIDIA-kernel-module-source-575.51.02/kernel-open'
make[2]: *** [Makefile:251: __sub-make] Error 2
make[2]: Leaving directory '/usr/lib/modules/6.14.5-zen1-1-zen/build'
make[1]: *** [Makefile:140: modules] Error 2
make[1]: Leaving directory '/tmp/tmp.2h3tBxgCGs/src/NVIDIA-kernel-module-source-575.51.02/kernel-open'
make: *** [Makefile:59: modules] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
  -> exit cleanup done
```

FuzzyQuils, 2025-05-15:
I made an attempt to patch the PKGBUILD myself to fix the build failures even though apparently the patches were being applied. https://github.com/Frogging-Family/nvidia-all/pull/291

Turns out there were a number of places where Kbuild and conftest.sh had to be patched during prepare() which took a while to figure out.

jamesb6626, 2025-05-27:
This is fixed now for me, not sure about @Martmists-GH. Probably ready to be closed if so.

closed 2025-05-27 by Martmists-GH