Frogging-Family/nvidia-all #293: 575.51.02 fails to install on linux-cachyos 6.15.0-2

z0n, 2025-05-27:
The DKMS build fails on Arch Linux with Kernel linux-cachyos 6.15.0-2:

```
DKMS (dkms-3.2.1) make.log for nvidia/575.51.02 for kernel 6.15.0-2-cachyos (x86_64)
Mon 26 May 23:11:02 CEST 2025

Building module(s)
# command: 'make' -j16 IGNORE_PREEMPT_RT_PRESENCE=1 NV_EXCLUDE_BUILD_MODULES='__EXCLUDE_MODULES' KERNEL_UNAME=6.15.0-2-cachyos modules LLVM=1
make[1]: Entering directory '/usr/lib/modules/6.15.0-2-cachyos/build'
make[2]: Entering directory '/var/lib/dkms/nvidia/575.51.02/build'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: clang version 19.1.7 (git@gitlab.archlinux.org:archlinux/packaging/packages/clang.git b536128bd29a83a26c0aafda19942995b30fc5fd)
  You are using:           clang version 19.1.7

Warning: Compiler version check failed:

The major and minor number of the compiler used to
compile the kernel:

clang version 19.1.7 (git@gitlab.archlinux.org:archlinux/packaging/packages/clang.git b536128bd29a83a26c0aafda19942995b30fc5fd), LLD 19.1.7

does not match the compiler used here:

clang version 19.1.7
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin

It is recommended to set the CC environment variable
to the compiler that was used to compile the kernel.

To skip the test and silence this warning message, set
the IGNORE_CC_MISMATCH environment variable to "1".
However, mixing compiler versions between the kernel
and kernel modules can result in subtle bugs that are
difficult to diagnose.

*** Failed CC version check. ***

  SYMLINK nvidia/nv-kernel.o
  SYMLINK nvidia-modeset/nv-modeset-kernel.o
 CONFTEST: set_pages_uc
 CONFTEST: list_is_first
 CONFTEST: set_memory_uc
 CONFTEST: set_memory_array_uc
 CONFTEST: set_pages_array_uc
 CONFTEST: ioremap_cache
 CONFTEST: ioremap_wc
 CONFTEST: ioremap_driver_hardened
 CONFTEST: ioremap_driver_hardened_wc
 CONFTEST: ioremap_cache_shared
 CONFTEST: pci_get_domain_bus_and_slot
 CONFTEST: get_num_physpages
 CONFTEST: pde_data
 CONFTEST: xen_ioemu_inject_msi
 CONFTEST: phys_to_dma
 CONFTEST: get_dma_ops
 CONFTEST: dma_attr_macros
 CONFTEST: dma_map_page_attrs
 CONFTEST: write_cr4
 CONFTEST: of_find_node_by_phandle
 CONFTEST: of_node_to_nid
 CONFTEST: of_get_ibm_chip_id
 CONFTEST: pci_stop_and_remove_bus_device
 CONFTEST: pci_rebar_get_possible_sizes
 CONFTEST: wait_for_random_bytes
 CONFTEST: register_cpu_notifier
 CONFTEST: cpuhp_setup_state
 CONFTEST: dma_map_resource
 CONFTEST: get_backlight_device_by_name
[... 339 lines cut ...]
      |          ^~~~~~~~~~~~~~~~
make[4]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/scripts/Makefile.build:208: nvidia/nv.o] Error 1
1 error generated.
nvidia/nv-dmabuf.c:24:10: fatal error: 'nv-dmabuf.h' file not found
   24 | #include "nv-dmabuf.h"
      |          ^~~~~~~~~~~~~
1 error generated.
make[4]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/scripts/Makefile.build:208: nvidia/nv-pci.o] Error 1
make[4]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/scripts/Makefile.build:208: nvidia/nv-dmabuf.o] Error 1
nvidia/nv-backlight.c:28:10: fatal error: 'os-interface.h' file not found
   28 | #include "os-interface.h"
      |          ^~~~~~~~~~~~~~~~
1 error generated.
make[4]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/scripts/Makefile.build:208: nvidia/nv-backlight.o] Error 1
make[3]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/Makefile:2039: .] Error 2
make[2]: *** [/usr/lib/modules/6.15.0-2-cachyos/build/Makefile:248: __sub-make] Error 2
make[2]: Leaving directory '/var/lib/dkms/nvidia/575.51.02/build'
make[1]: *** [Makefile:248: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.15.0-2-cachyos/build'
make: *** [Makefile:140: modules] Error 2

# exit code: 2
# elapsed time: 00:00:18
----------------------------------------------------------------
```

z0n, 2025-05-28:
Ah, too bad. Went for the CachyOS open drivers for now as I don't have the old kernel anymore 😅 CachyOS doesn't have package archives, does it?

Thanks for your work 🫡

ptr1337, 2025-05-29:
> Ah, too bad. Went for the CachyOS open drivers for now as I don't have the old kernel anymore 😅 CachyOS doesn't have package archives, does it?
> 
> Thanks for your work 🫡

We do, but only for 3 versions:
https://archive.cachyos.org/archive/

blackPantherOS, 2025-06-03:
EXTRA_CFLAGS values does not work in 6.15.x 
sed -i 's|EXTRA_CFLAGS |KBUID_CFLAGS |g' /path/to/NVIDIA-SOURCE/Kbuild

closed 2026-08-19 by damachine